ioos/glider-dac #365: Fix the Global attribute dac_qc_comment to list clear messages about variable issues

leilabbb, 2024-08-13:
This is related to #318

The dac_qc_comment was added to the file's global attributes to list issues encountered when implementing the QARTOD tests.

- [ ] The issues related to non-standard standard_name and the use of standard standard_name by more than one variable need to be rewritten to convey a clear message.  Click the related [code](https://github.com/ioos/glider-dac/blob/main/glider_qc/glider_qc.py#L160-L169)

A [dataset](https://gliders.ioos.us/erddap/tabledap/sp001-20150111T2247.html) with an example of the issue: 

> dac_qc_comment "QC skipped **forsea_water_practical_salinity**: _the same standard name is shared by_ , Skipped QC: density is an array of nans";

- [ ] The issues related to empty arrays need to be edited for clarity. Click the related [code](https://github.com/ioos/glider-dac/blob/main/glider_qc/glider_qc.py#L393-L400)

A [dataset](https://gliders.ioos.us/erddap/tabledap/sp066-20171129T1616.html) with an example of the issue:

> dac_qc_comment "Skipped QC: temperature is an array of **nansSkipped** QC: conductivity is an array of **nansSkipped** QC: density is an array of nansSkipped QC: pressure is an array of **nansSkipped** QC: salinity is an array of nans";

sarinamann-noaa, 2025-02-12:
Issue itself is resolved, but changes are not reflected in ALL files in ERDDAP yet. Waiting to close until changes are all reflected, like with the new update.

sarinamann-noaa, 2025-05-15:
@leilabbb can this issue be closed out?

leilabbb, 2025-05-16:
> [@leilabbb](https://github.com/leilabbb) can this issue be closed out?

This issue needs to remain open until we find a solution to have the dac_qc_comment global variable show more than just the comments from the latest QCed netCDF file found in the directory. This is also related to issue #408.