ioos/glider-dac #366: Update documentation with a section about QARTOD implementation

leilabbb, 2024-08-14:
This is related to issue #355 

Update the [documentation](https://github.com/ioos/glider-dac/blob/8fc8ff41be4ef585a7f7f7bd64ece2dc19e28384/_docs/ngdac-netcdf-file-format-version-2.md) with information about the QC process: 

- [ ] GDAC implementation of QARTOD flags for real-time data 
- [ ] Data providers' requirements to submit QC results to the DAC using a standardized QC variable structure and format.

sarinamann-noaa, 2024-10-02:
Will discuss at GDAC meeting 10/02/2024

leilabbb, 2026-09-09:
please review the proposed changes in [Commit 4053948]( https://github.com/ioos/glider-dac/commit/4053948964e77a9975b995048b413e02202d6300)

leilabbb, 2026-09-17:
changes to sidebar menu and documentation is in Pull Request #648

leilabbb, 2026-10-01:
closing issue: netcdf-qartod-gdac-what-to-expect.md  is now integrated into the wiki pages.

closed 2026-10-01 by leilabbb