ioos/glider-dac #368: WMO ID not appearing correctly on the providers page.

leilabbb, 2024-08-19:
Example of the [navoceano operator](https://gliders.ioos.us/providers/operators/NAVOCEANO/deployments): 

There are cases with WMO ID duplicated: 
   see Deployment ng222-20210929T0000 WMO ID 48029844802984
   The deployment on [ERDDAP](https://gliders.ioos.us/erddap/info/ng222-20210929T0000/index.html?platform_meta) shows the correct wmo_id = 4802984

There are cases with no WMO ID: 
   see Deployment ng1050-20240717T0000 WMO ID  null
   The deployment on [ERDDAP](https://gliders.ioos.us/erddap/info/ng1050-20240717T0000/index.html) shows the correct  wmo_id = 8901072

sarinamann-noaa, 2024-10-02:
If this is fixed, request @benjwadams to close this issue

leilabbb, 2025-01-29:
The case with no WMO ID for [deployment ng1050-20240717T0000](https://gliders.ioos.us/providers/operators/NAVOCEANO/deployments) is **FIXED** with the expected wmo_id = 8901072 

The case for [deployment ng222-20210929T0000](https://gliders.ioos.us/providers/operators/NAVOCEANO/deployments) with repeated WMO ID is still an issue.

sarinamann-noaa, 2025-03-17:
@leilabbb is there an update on this issue?

leilabbb, 2025-03-18:
> [@leilabbb](https://github.com/leilabbb) is there an update on this issue?
This issue is under investigation.

benjwadams, 2025-04-03:
For the excessively long WMO IDs, I've truncated these strings down to seven characters.

https://github.com/ioos/glider-dac/pull/424/commits/bf0148708525205015dcd3755c8678b49015d7e4 also prevents the `wmo_id` field from exceeding seven characters in length, but will need to be merged from `develop` in order to take effect.

benjwadams, 2025-04-09:
@leilabbb, can this be closed in light of recent changes made above?

sarinamann-noaa, 2025-05-15:
@leilabbb can this issue be closed?

leilabbb, 2025-05-16:
> For the excessively long WMO IDs, I've truncated these strings down to seven characters.
> 
> [bf01487](https://github.com/ioos/glider-dac/commit/bf0148708525205015dcd3755c8678b49015d7e4) also prevents the `wmo_id` field from exceeding seven characters in length, but will need to be merged from `develop` in order to take effect.

closing this issue, the wmo_id repetition has been resolved.

closed 2025-05-16 by leilabbb