ioos/glider-dac #373: problem reading salinity and density values - why are the values masked ?

leilabbb, 2024-08-30:
The [QARTOD flags for salinity and density](https://gliders.ioos.us/erddap/tabledap/ru39-20230817T1520.html) aren’t displaying correctly because the netCDF4 library isn’t reading the values properly (all values are masked). This issue doesn't happen when using xarray or ncdump to read the variables. For details, see the [notebook](https://gist.github.com/leilabbb/84c09d8133d3ef89d625807ee5c34199).

This is relevant to #274

benjwadams, 2024-09-03:
This issue resides on the provider side.  `valid_min` and `valid_max` are swapped in the dataset, resulting in no data being within the valid range.   By default, Python-NetCDF4 automatically masks this data.

<class 'netCDF4._netCDF4.Variable'>
float32 density(time)
    _FillValue: 9.96921e+36
    instrument: instrument_ctd
    long_name: Density
    observation_type: calculated
    platform: platform
    processing_level: 2
    standard_name: sea_water_density
    units: kg m-3
    valid_max: 1015.0
    valid_min: 1040.0
    ancillary_variables: instrument_ctd qartod_density_flat_line_flag qartod_density_gross_range_flag qartod_density_rate_of_change_flag qartod_density_spike_flag qartod_density_primary_flag
unlimited dimensions: time
current shape = (43,)

That said, compliance-checker implements valid_min/max and valid_range checks, but we haven't integrated them into the Glider DAC results yet.  It's definitely something we could use in future compliance checker versions.

benjwadams, 2024-10-14:
Closed as provider issue.  Potential follow up includes changing metadata or having it reuploaded.

closed 2024-10-14 by benjwadams