ioos/glider-dac #374: User Request: Expanded availability of rsync

sarinamann-noaa, 2024-08-30:
Jody Klymak (jklymak@uvic.ca) via Slack:
"any chance rsync will become available?  Will save a lot of duplicate transfers for our group"

Response:
"We will create a GitHub issue to expand rsync use and consider it when we are expanding capabilities. Feel free to reach out to the GDAC support team or contact me directly if you would like more detailed information on the process."

benjwadams, 2024-09-03:
Giving rsync access means giving SSH access on some level so we'd have to clear this with IT at the very least.

Also, in (non-SFTP) FTP, the users exist as virtual users in a database.  Rsync/SSH would require setting up individual users on the system, unless some other mechanisms for authentication exist that I'm not aware of.

sarinamann-noaa, 2025-07-02:
- Ben intended to work with Jennifer Sevadjian (University of California, San Diego) to remedy missing file and assess the group as an rsync candidate.
- The user request ended up being more complex than previously thought- permissioning failed and requires additional work/troubleshooting. GDAC is unable to provide complete support/rsync access for Jenn at this time.
- This process could be beneficial to all users, but has to be handled on a case-by-case basis. GDAC plans to continue exploring this when we are out of "mission-critical" restrictions.

benjwadams, 2025-07-09:
IT has put the kibosh on this:

> no direct rsync activity between untrusted domains allowed per Security. Please use an ftp site to xfer files instead

Closing as unplanned.