ioos/glider-dac #376: rate_of_change_test flags are corrupted when the time array is in nano seconds

leilabbb, 2024-09-04:
The QARTOD rate_of_change_test expects time in seconds. When the time array uses a different data type the array needs to be tested and a conversion needs to be applied for consistency.

The time data type also affects the threshold calculation in the [glider_qc.py](https://github.com/ioos/glider-dac/blob/main/glider_qc/glider_qc.py#L185-L198). An alternative method is tested to render a more reliable value for the threshold. 

This [notebook](https://gist.github.com/leilabbb/847cf9ccaf562edd65abaddb00fae26d) tests the modifications needed for the threshold calculation and applies them to a NetCDF file.

Data annotation will be added to the file to notify users. This is part of the issue https://github.com/ioos/glider-dac/issues/365

benjwadams, 2024-09-05:
Nanoseconds aren't accepted by ERDDAP, either.  May be an issue because it appears to be valid UDUnits.

leilabbb, 2024-11-13:
A [draft pull request ](https://github.com/ioos/glider-dac/compare/main...qc_fixes#diff-652f8f887a8fc6b1ff6f471c7266a1a7e559612431b158345db606a595869fd7R416) with a new conditional statement was created to check time units .

```
if times.dtype == "timedelta64[ns]":
    log.info("timestamps are in nanoseconds")
    report += "| Invalid timestamps: unit in nanoseconds"
```

closed 2024-12-24 by benjwadams