ioos/glider-dac #380: Develop a plan for GDAC QC

kbailey-noaa, 2024-10-02:
### What can be improved? What is missing?

FY24 GDAC SOW: 
Per IOOS Certification requirements, data served via the GDAC must be quality controlled, whether by routines applied by the provider or by the GDAC. The GDAC must apply QARTOD to variables that have existing QARTOD manuals (e.g. [Manual for QC of Glider Temperature and Salinity Data](https://cdn.ioos.noaa.gov/media/2017/12/Manual-for-QC-of-Glider-Data_05_09_16.pdf)).  Flags must be published in the data files following IOOS Metadata standards ([ioos.github.io](https://ioos.github.io/ioos-metadata/ioos-metadata-profile-v1-2.html#quality-controlqartod)). 

The GDAC Team shall develop a plan for full implementation of QC in the GDAC using the following tiered approach:
**Near-term**: implement or repair existing Required and Strongly Recommended QARTOD tests, using global thresholds for all GDAC data, and publish aggregate QC flags (qc_agg).
**Mid-term**: Use QARTOD manuals and standards to understand and identify the potential for improvements to existing tests (e.g. regional refinement of thresholds, additional real-time tests, etc).  
**Long-term**: Engage a scientific working group, possibly under UG2, to investigate the potential for delayed-mode quality control and/or to gain feedback on additional QC considerations.   

1) Document an implementation plan that addresses the above 3 aspects of QC, which includes what we're doing now, gaps that remain, and what we intend to do, for both real-time and delayed mode datasets.  Consider this plan to be a proposal that we'd run by the community.

2) Identify a forum to document this plan. I highly discourage Word documents, since those aren't interactive. Options are the GitHub wiki (I can activate this), some type of Slack feature (?), the GitHub Discussions option...something else?

leilabbb, 2024-12-13:
The **Near-Term** goal has been implemented in the GDAC: 

About implementing or repairing existing required and/or strongly recommended QARTOD tests:

:: Five geophysical datasets are quality-controlled using the IOOS-QC QARTOD modules.
Temperature
Conductivity
Density
Pressure
Salinity

:: Five test functions are used to perform quality control on the geophysical variables.
Gross Range Test
Spike Test
Rate of Change Test
Flat Line Test
Aggregate Quality Flag

:: The QARTOD location test is customized to align with the parameters available in the gliders' profile files. The two variables used in the location test are:
Longitude
Latitude

About using global thresholds for all GDAC data.
:: The GDAC uses a configuration file to set up the global thresholds for the QARTOD tests. 

:: The thresholds for the spike test and the rate of change test are updated during the QC process to align with the datasets' ranges.

About publishing aggregate QC flags (qc_agg).
:: The flags are calculated using the QARTOD compare function, which employs a prioritization technique to generate a final list of quality flags for a geophysical variable..
priority list:
     {
        QartodFlags.MISSING,
        QartodFlags.UNKNOWN,
        QartodFlags.GOOD,
        QartodFlags.SUSPECT,
        QartodFlags.FAIL,
    }

leilabbb, 2024-12-22:
About the **long-term** goal: 

`Engage a scientific working group, possibly under UG2, to investigate the potential for delayed-mode quality control and/or to gain feedback on additional QC considerations.`

 - With input from the glider community, this effort will provide a roadmap to improve and guide the GDAC QC application, enhancing its ability to better serve the community.
 - An open issue (#355) regarding delayed-mode datasets has initiated a conversation that can be shared with the community for further discussion.
 - Additionally, issue (#274) highlights critical aspects of the GDAC QC application.