ioos/glider-dac #385: Completed missions not passing compliance checker due to qartod variables

jenseva, 2024-10-23:
### What went wrong?

Hi There,  

When checking on our completed missions to see if they have successfully been sent to NCEI I now see errors related to DAC supplied QARTOD variables.

On page [https://gliders.ioos.us/providers/users/jpsevadjian/deployment/64fcec6dbcab6800096055f3](https://gliders.ioos.us/providers/users/jpsevadjian/deployment/64fcec6dbcab6800096055f3), I see:

Compliance Check Report
CF standard name errors must be fixed prior to NCEI archival:

Attribute long_name or/and standard_name is highly recommended for variable qartod_monotonic_pressure_flag

Another mission page ([https://gliders.ioos.us/providers/users/jpsevadjian/deployment/660ef6ab24c258000aec233c](https://gliders.ioos.us/providers/users/jpsevadjian/deployment/660ef6ab24c258000aec233c)) shows an additional error: 

Compliance Check Report
CF standard name errors must be fixed prior to NCEI archival:

Attribute long_name or/and standard_name is highly recommended for variable qartod_monotonic_pressure_flag
Attribute long_name or/and standard_name is highly recommended for variable qartod_location_flag

I believe these are issues to be resolved by the DAC implementation of QARTOD and are similar to those reported in [issue 382](https://github.com/ioos/glider-dac/issues/382). 

Should we be concerned about these errors and are they preventing our data from being submitted to NCEI?

Thanks,
Jenn

### Operating System

macOS

### Version

12.7.1

### Python Version

3.9

### Code to Reproduce

```python
None. Visit https://gliders.ioos.us/providers/users/jpsevadjian/deployment/660ef6ab24c258000aec233c](https://gliders.ioos.us/providers/users/jpsevadjian/deployment/660ef6ab24c258000aec233c to see the message.
```

### Errors, Traceback, and Logs

```pytb
Compliance Check Report
CF standard name errors must be fixed prior to NCEI archival:

Attribute long_name or/and standard_name is highly recommended for variable qartod_monotonic_pressure_flag
Attribute long_name or/and standard_name is highly recommended for variable qartod_location_flag
```

leilabbb, 2024-10-25:
working on fixing the [ERDDAP catalog variables information](https://github.com/ioos/glider-dac/blob/0685f728ffa41dce336a32f7577b9dd7e027fd86/scripts/build_erddap_catalog.py#L649-L684) to avoid the missing attribute long_name or/and standard_name

sarinamann-noaa, 2024-11-04:
@jenseva 
Providing an update from the GDAC team: 
The pull request was merged, and changes have been applied but it will take sometimes to see them applied to datasets.

Please feel free to reach out for more information- thank you

leilabbb, 2024-11-08:
Changes are now seen on ERDDAP. Checkout:

[sp028-20230901T1905](https://gliders.ioos.us/erddap/tabledap/sp028-20230901T1905.html) dataset' QARTOD variables all include standard names and long names.

[sp020-20240403T1710](https://gliders.ioos.us/erddap/tabledap/sp020-20240403T1710.html) dataset' QARTOD variables all include standard names and long names. 

A rerun of the compliance checker should reflect those fixes.

leilabbb, 2024-11-13:
The errors are still showing up in the compliance checker report. The ticket will be closed once the report is updated.
> Compliance Check Report
> CF standard name errors must be fixed prior to NCEI archival:
> 
> Attribute long_name or/and standard_name is highly recommended for variable qartod_monotonic_pressure_flag
> Attribute long_name or/and standard_name is highly recommended for variable qartod_location_flag

benjwadams, 2024-11-27:
Fixed after #399, https://github.com/ioos/glider-dac/commit/912a258f8f1aa50e73aa461c18c620885329861b and resynced submitted files, closing as completed.

closed 2024-11-27 by benjwadams