ioos/glider-dac #390: QARTOD test time casting fails

benjwadams, 2024-11-05:
### What went wrong?

Certain datasets attempt to run QARTOD and a cast to timestamp subsequently fails:

```
/usr/local/lib/python3.8/site-packages/xarray/coding/times.py:250: RuntimeWarning: invalid value encountered in cast
  flat_num_dates_ns_int = (flat_num_dates * _NS_PER_TIME_DELTA[delta]).astype(
```

May be due to attempting to cast floating point using integer conversion methods, see below.

### Operating System

Linux

### Version

Production

### Python Version

3.8 per production

### Code to Reproduce

```python
Running `ncdump -t time /data/data/priv_erddap/OOI-CGSN/cp_380-20240821T1404/cp_380_20240918T144043Z_rt.nc` results in these time variables being returned:

 time = 1726670443.45795, 1726670504.53198, 1726670564.5809, 
    1726670624.61365, 1726670684.68829, 1726670744.73468, 1726670804.77075, 
    1726670864.81549 ;
}
```

### Errors, Traceback, and Logs

```pytb
As above.
```

leilabbb, 2024-11-20:
This is related to _xarray_ decoding time into _timedelta64[ns]_ values. If any of the decoded values used in arithmetic operations produce _NaT_, _NaN_, or _inf_ values due to the long series of numbers after the decimal point (e.g., '2024-09-18T14:40:43.457950208'), it results in an invalid value warning.

This warning should not affect the QC process and  could be ignored using the warnings package. See [documentation](https://docs.python.org/3/library/warnings.html#warning-filter).

benjwadams, 2024-11-26:
Issue is minor enough not to warrant keeping open, but consider suppressing warnings if desired.

closed 2024-11-26 by benjwadams