ioos/glider-dac #393: Broken Link- "DAC Specifications"

sarinamann-noaa, 2024-11-07:
Reading through an email addressing a different issue submitted by a user, I clicked this link: 
https://gliders.ioos.us/status/?cat=datasets&mode=rt&missing=True&status=all

And the following part of the error message contains a broken link:
"Submitted NetCDF files do not meet the [DAC specification](https://ioos.github.io/ioosngdac/ngdac-netcdf-file-format-version-2.html)" this link leads to a 404 error. I am unsure if other pages also point to this broken link.

leilabbb, 2024-11-08:
The [DAC specification](https://ioos.github.io/ioosngdac/ngdac-netcdf-file-format-version-2.html) is pointing to an [archived repository](https://github.com/ioos/ioosngdac). 
A replacement should be pointing to the current GDAC repository [netCDF templates folder](https://github.com/ioos/glider-dac/tree/gh-pages/_nc/template) with the latest version [IOOS_Glider_NetCDF_v3.0-qartod](https://github.com/ioos/glider-dac/blob/gh-pages/_nc/template/IOOS_Glider_NetCDF_v3.0-qartod.cdl) which is out of date.

We need to create version 4 and update to documentation and associated links.

leilabbb, 2026-09-17:
OBE -  the status page is no longer in use and will be replaced.

closed 2026-09-17 by leilabbb