ioos/glider-dac #394: ERDDAP Upgrade to 2.25.1

sarinamann-noaa, 2024-11-14:
**The GDAC version of ERDDAP needs to be upgraded**. This is a priority immediately after hurricane season. Once completed, this should be reflected in the GDAC release notes.

Please see the information sent in the ERDDAP Google Group below:
ERDDAP v2.25.1 as been released.  It has a number of new features,  including zarr read support as well as parquet read and write support.  Please update your ERDDAP if at all possible,  this insures that all libraries used in ERDDAP are the most recent patched versions. Also keep your apache/nginx/tomcat/java up-to-date.

For Matlab users,  a package that has the same functionality as the R package 'rerddapXtracto'   can be found at https://github.com/rmendels/xtractoMatlab

sarinamann-noaa, 2024-11-14:
https://github.com/axiom-data-science/docker-erddap/pull/91

benjwadams, 2024-12-18:
Awaiting release of ERDDAP Docker version, and will update after release.

benjwadams, 2025-01-14:
Should be available now as https://github.com/axiom-data-science/docker-erddap/pull/96 supersedes previously linked PR as long as the image is available.

benjwadams, 2025-02-13:
Closed by #405

closed 2025-02-13 by benjwadams