ioos/glider-dac #395: Filter Out Invalid Datasets at Submission

leilabbb, 2024-11-15:
### What should we add?

Add filtering for datasets that are out of compliance with the latest [IOOS_Glider_NetCDF version](https://github.com/ioos/glider-dac/tree/gh-pages/_nc/template) to prevent failure when loading into ERDDAP.

### Reference

The GliderDAC Compliance Checker Plugin generates a [report](https://compliance.ioos.us/report/706cf2b6b639526cb1a1204cdbfd8185012d8ede) detailing any existing issues and recommended corrective actions, in accordance with the [NGDAC NetCDF File Format Version 2](https://github.com/ioos/ioosngdac/wiki/NGDAC-NetCDF-File-Format-Version-2).

We can use this tool to enhance the report capabilities, determine whether a dataset passes or fails, and avoid ERDDAP errors, such as the case with the dataset sg658-20220408T0000 reported by Amber York.

benjwadams, 2024-12-06:
We already do run the Glider Compliance Checker against datasets, and the whole IOOS Gliders spec is used, however, we only actually apply the standard names section of the compliance report to see if the aggregation can be archived by NCEI.  We had tried running against the full spec, but if I recall correctly, currently *no* dataset passes this.

sarinamann-noaa, 2025-02-12:
Issue replaced with https://github.com/ioos/glider-dac/issues/415

closed 2025-03-18 by leilabbb