ioos/glider-dac #406: Update version to 2.0

kbailey-noaa, 2025-01-28:
Please update https://github.com/ioos/glider-dac/releases/tag/v1.0 to v2.0.  The v2.0 should include a brief bulleted list of the major changes that occurred since 2021.

leilabbb, 2025-01-29:
Here the link to version 2 draft:
https://github.com/ioos/glider-dac/releases

kbailey-noaa, 2025-01-29:
Thanks! I see these are all specific to QARTOD / QC updates.  Can you (the team) include additional items, such as release of a [Dev environment](https://github.com/ioos/glider-dac/issues/232), etc?  Has anything major happened wrt archival?  Data ingestion?
I also recommend a couple generic additional bullets/items such as:
-- Bug fixes to the GDAC Map
-- Security upgrades
-- ERDDAP updates
(I'm sure you have better wording, but hopefully you get the idea). 

The point is just to indicate to the community that there are various elements of the system that are within focus, in addition to QC.

kbailey-noaa, 2025-02-21:
@sarinamann-noaa FYI I went to the main GDAC github page (https://github.com/ioos/glider-dac) and hit the "+1 release" link and I see the updates here: https://github.com/ioos/glider-dac/releases
Then I hit the 'edit' button it led me here: https://github.com/ioos/glider-dac/releases/edit/untagged-ce10073477818e4f1403 where the proposed changes are laid out.

@leilabbb @benjwadams Can you spell out the word version so it's "Version 2.0" instead of v2.0, for consistency? The content looks good otherwise.  Then please publish.

closed 2025-04-03 by benjwadams