ioos/glider-dac #408: Missing QARTOD flag variable results

kerfoot, 2025-02-04:
I searched through the open issues but did not find anything specific to this, so I'm filing a new issue. I was checking the QARTOD results for a glider that we currently have in the water [maracoos_06-20250116T1522](https://gliders.ioos.us/erddap/tabledap/maracoos_06-20250116T1522.html).

The following qartod variables are filled entirely with NaNs/_FillValue:

+ qartod_density_flat_line_flag
+ qartod_density_gross_range_flag 
+ qartod_density_primary_flag
+ qartod_density_rate_of_change_flag
+ qartod_density_spike_flag
+ qartod_salinity_flat_line_flag
+ qartod_salinity_gross_range_flag 
+ qartod_salinity_primary_flag
+ qartod_salinity_rate_of_change_flag
+ qartod_salinity_spike_flag

This appears to be the case for many of the active real-time data sets on the DAC ERDDAP server.

Does this mean that the these tests were not run by the DAC? If not, what does it mean? I would assume that, at a minimum, the *\*salinity\** flags are required by NDBC to omit bad salinity values, but they are not available.

kerfoot, 2025-02-26:
I see that this has been moved to the "No Status" group on the project management page. Wondering what that means? Is there any clarity on this issue moving forward with respect to why the checks are not being run and how it will be fixed?

leilabbb, 2025-03-06:
This issues is under review. 
FYI: maracoos_06-20250116T1522 has many files with issues such as:
- duplicates timestamps 
- switched valid_min and valid_max 
- not enough valid data for a QC test run.  
-  etc.

benjwadams, 2025-04-09:
For the QARTOD tests that are NAN'ed out, the `valid_min` and `valid_max` are set to values that are possibly swapped and impossible to fulfill:

```
        float density(time) ;
                density:_FillValue = 9.96921e+36f ;
                density:ancillary_variables = "instrument_ctd" ;
                density:instrument = "instrument_ctd" ;
                density:long_name = "Density" ;
                density:observation_type = "calculated" ;
                density:platform = "platform" ;
                density:processing_level = 2 ;
                density:standard_name = "sea_water_density" ;
                density:units = "kg m-3" ;
                density:valid_max = 1015. ;
                density:valid_min = 1040. ;
```

The QARTOD process actually catches this issue and denotes it in the logs:

```
gliderdac: glider_qc.glider_qc.qc_task('/data/data/priv_erddap/rutgers/maracoos_06-20250116T1522/maracoos_06_20250127T053843Z_rt.nc', '/glider-dac/data/qc_config.yml') (676b8af1-dd5d-4626-88fe-e8554fec0b41)
Loading config from /glider-dac/data/qc_config.yml
 Running IOOS QARTOD tests on maracoos_06_20250127T053843Z_rt.nc
Found 5 variables for QARTOD tests: ['temperature', 'conductivity', 'density', 'pressure', 'salinity']
Created 5 QC Variables for temperature
Insufficient data: both 'values' and 'times' must have at least two elements.
Generated QC test results for temperature
Updating qartod_temperature_gross_range_flag
Updating qartod_temperature_spike_flag
Updating qartod_temperature_flat_line_flag
Updating qartod_temperature_primary_flag
Created 5 QC Variables for conductivity
Insufficient data: both 'values' and 'times' must have at least two elements.
Generated QC test results for conductivity
Updating qartod_conductivity_gross_range_flag
Updating qartod_conductivity_spike_flag
Updating qartod_conductivity_flat_line_flag
Updating qartod_conductivity_primary_flag
Created 5 QC Variables for density
density: valid_min (1040.0) and valid_max (1015.0) are switched
density: valid_min (1040.0) and valid_max (1015.0) are switched
Created 5 QC Variables for pressure
Insufficient data: both 'values' and 'times' must have at least two elements.
```

leilabbb, 2025-04-16:
Adding to the discussion above...

What you see in ERDDAP as far as variables and attributes is directly related to the latest netCDF file found in the directory (see **Example** below). This is expected since it is coded in [build_erddap_catalog.py Line#L839-L848](https://github.com/ioos/glider-dac/blob/main/scripts/build_erddap_catalog.py#L839-L848) 

We are working on creating a new logic to be more consistent with what should be displayed based on all files combined. 

**Example:**

**Deployment**: maracoos_06-20250116T1522
**Latest file read**: maracoos_06_20250210T213253Z_rt.nc

**Variable**: qartod_temperature_flat_line_flag
>  {
    Byte _FillValue 2;
    String _Unsigned "false";
    Byte actual_range 1, 9;
    String dac_comment "**QARTOD TESTS NOT RUN**";
    String flag_meanings "PASS NOT_EVALUATED SUSPECT FAIL MISSING";
    Byte flag_values 1, 2, 3, 4, 9;
    String ioos_category "Quality";
    String long_name "QARTOD Flat Line Test for sea_water_temperature";
    String references "https://gliders.ioos.us/files/Manual-for-QC-of-Glider-Data_05_09_16.pdf";
    String standard_name "flat_line_test_quality_flag";
    String units "1";
    Byte valid_max 9;
    Byte valid_min 1;
   }

**Global attribute**: dac_qc_comment
>  "maracoos_06-20250116T1522 (maracoos_06_20250210T213253Z_rt.nc: **no variable found with sea_water_temperature** no variable found with sea_water_electrical_conductivity no variable found with sea_water_density no variable found with sea_water_practical_salinity  spike_test dropped for pressure: Not enough valid data for std calculation. rate_of_change_test dropped for pressure: Not enough valid data points for std and mean calculations.)"