ioos/glider-dac #420: Allow metadata modifications

benjwadams, 2025-02-13:
**User Story:**
As a user or admin, I want to be able to modify NetCDF metadata attributes in my own deployments (or anyone's, if I'm an admin)

Right now we can modify metadata attributes by uploading an `extra_atts.json` file with either the variable name or a key denoting global metadata attributes.   Ideally, this functionality would be accessible to users who uploaded or are affiliated with an organization with a deployment from the deployment page itself once logged in.  The data attributes would be persisted to a key value pair table joined one to many against a deployment and metadata modifications would also have provenance information through the modification of the NetCDF `history` attribute.  The existing `extra_atts.json` mechanism would still be usable via an API call through a `PUT` request, but would no longer reside on the filesystem in the deployment directory.  The existing mechanism with the JSON file is often used for CF Standard Name alterations to pass the CF Compliance Check prior to archival at NCEI.  This functionality would continue to exist through modifications made via the Providers App deployment page view as well as API calls for more advanced users.   There are other potential uses as well, such as setting the `gts_ingest` attribute to determine which attributes to consume for downstream data consumers, as noted in #391.

kerfoot, 2025-07-24:
As a follow up to this, I have a question about an old (completed) deployment that I need to modify the sea_name attribute as requested by the DAC administrators. The deployment is [ru38-20250414T1500](https://gliders.ioos.us/providers/users/rutgers/deployment/67fdb9394ec6d00008682a2c). I need to change the sea_name to "Gulf of America". I would like to do this using the extra_atts.json file, but the deployment has already been marked as completed. Here are my questions:

1. Are changes noted in the extra_atts.json file propagated to the datasets.xml file if the deployment has been marked as complete and archived for NCEI? 
2. If not, it is my understanding that, once a deployment has been marked as complete and the archival package created for NCEI, any changes to files do not trigger a re-archival process. So this means that I cannot re-process the data set and upload the new files in order to have the changes take effect in both the ERDDAP data set and NCEI archival package?
3. What is the preferred way to make changes to a completed deployment and have them included in the NCEI archival package and ERDDAP data set?

benjwadams, 2025-07-29:
1) Adding/updating extra_atts.json should update the modified field for the deployment.  This, in turn, should be picked up by build_erddap_catalog.py script.

2) The aggregation goes to "public" ERDDAP folder.  This is a historical feature no longer used since we fixed aggregation speed some time ago.  It could entirely be moved to archival instead.   However, unlike the previous step, the public ERDDAP aggregation is only recreated when a *netCDF* file has newly been detected as modified. Archival is hard linked against files in the public ERDDAP, so the change in the archived file gets picked up since the file checksum will have changed.

3) Right now, touch a file and reupload it after having submitted extra_atts.json.  This workflow could be improved by using the database as the source of truth.

I'd recommend that a separate issue be opened and these details included.  The issue here is more of a user interface change and mechanism change (namely, moving away from extra_atts.json file, although it'll still be supported for backwards compatibility.)

benjwadams, 2025-07-31:
Changes were loaded, but required some manual intervention in the form of issuing hardFlag refreshes.  Further discussion can be had in a separate issue, but we may want to use ERDDAP's subscription service to better signal changes.

leilabbb, 2026-11-05:
In addition to the capability of adding an extra_atts.json file to the deployment folder, there is the option to have a web field on the providers page that takes a json formatted text entry to correct and modify metadata. This will probably help providers submit a well formatted json file that the system can read.