ioos/glider-dac #425: Broken IMOS Glider Documentation Link in gh-pages branch

benjwadams, 2025-02-20:
### What went wrong?

In `gh-docs` branch, there's a dead link referring to http://imos.org.au/fileadmin/user_upload/shared/ANFOG/ANFOG_DataManagement_UsersManual_v5.1_21Aug2018.pdf
which is now 404 and reported by the markdown link checker: https://github.com/ioos/glider-dac/blob/gh-pages/_docs/ngdac-netcdf-file-format-version-2.md?plain=1#L51

I tried to find this document and was only able to find the most recent version on archive.org.  It looks like it may have possibly been superseded by https://repository.oceanbestpractices.org/bitstream/handle/11329/1030/Delayed_Mode_QAQC_Best_Practice_Manual_OceanGliders_v2.0.pdf?sequence=1

### Operating System

Windows

### Version

N/A

### Python Version

N/A

### Code to Reproduce

```python
#
```

### Errors, Traceback, and Logs

```pytb
#
```

ocefpaf, 2025-02-21:
> I tried to find this document and was only able to find the most recent version on archive.org. It looks like it may have possibly been superseded by https://repository.oceanbestpractices.org/bitstream/handle/11329/1030/Delayed_Mode_QAQC_Best_Practice_Manual_OceanGliders_v2.0.pdf?sequence=1

I can send a PR using that link, but it would be nice to get a confirmation from the community if that is the correct one.

benjwadams, 2025-02-24:
Awaiting reply from IMOS.

In the interim, we could possibly use the archive.org copy: https://web.archive.org/web/20220621043652/http://imos.org.au/fileadmin/user_upload/shared/ANFOG/ANFOG_DataManagement_UsersManual_v5.1_21Aug2018.pdf

leilabbb, 2025-02-25:
> It looks like the above link is also the document I had thought would replace the old manual, albeit a more recent version.

Yes the link from IMOS is the correct replacement.

benjwadams, 2025-02-25:
Fixed by #426.

closed 2025-02-25 by benjwadams