ioos/glider-dac #438: OS X/ARM test runners

benjwadams, 2025-03-27:
### What should we add?

@leilabbb is having issues building Glider DAC against the `develop` branch due to UDUnits not finding certain files such as the udunits.xml file.  She's running an ARM architecture Mac as far as I know. I don't encounter the same issues building against Linux/x86_64, and it appears that neither does GitHub Actions presently.

While fixing UDUnits on multiple architectures would be helpful, it'd also be good to have test runners against OS X ARM architectures.

leilabbb, 2025-03-27:
Apple Silicon (ARM): arm64 
Git branch: develop

command: `docker-compose up -d `

Error output text:
> 50.45   Building wheel for cf-units (pyproject.toml): started
50.59   Building wheel for cf-units (pyproject.toml): finished with status 'error'
50.59   error: subprocess-exited-with-error
50.59   
50.59   × Building wheel for cf-units (pyproject.toml) did not run successfully.
50.59   │ exit code: 1
50.59   ╰─> [22 lines of output]
50.59       Traceback (most recent call last):
50.59         File "/usr/local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
50.59           main()
50.59         File "/usr/local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
50.59           json_out["return_val"] = hook(**hook_input["kwargs"])
50.59                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
50.59         File "/usr/local/lib/python3.12/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 280, in build_wheel
50.59           return _build_backend().build_wheel(
50.59                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
50.59         File "/tmp/pip-build-env-k7a184ep/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 438, in build_wheel
50.59           return _build(['bdist_wheel', '--dist-info-dir', str(metadata_directory)])
50.59                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
50.59         File "/tmp/pip-build-env-k7a184ep/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 426, in _build
50.59           return self._build_with_temp_dir(
50.59                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
50.59         File "/tmp/pip-build-env-k7a184ep/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 407, in _build_with_temp_dir
50.59           self.run_setup()
50.59         File "/tmp/pip-build-env-k7a184ep/overlay/lib/python3.12/site-packages/setuptools/build_meta.py", line 320, in run_setup
50.59           exec(code, locals())
50.59         File "<string>", line 156, in <module>
50.59         File "<string>", line 76, in get_package_data
50.59       ValueError: Can't open UDUNITS2_XML_PATH file /usr/share/xml/udunits during cf-units wheel build.
50.59       [end of output]
50.59   
50.59   note: This error originates from a subprocess, and is likely not a problem with pip.
50.59   ERROR: Failed building wheel for cf-units
50.59   Building wheel for sgmllib3k (pyproject.toml): started
50.67   Building wheel for sgmllib3k (pyproject.toml): finished with status 'done'
50.67   Created wheel for sgmllib3k: filename=sgmllib3k-1.0.0-py3-none-any.whl size=6091 sha256=f7edc50883b5c0dc2e89440d243f483f404f8a37e54a48c83eb29e33611c34df
50.67   Stored in directory: /tmp/pip-ephem-wheel-cache-ok__r__3/wheels/03/f5/1a/23761066dac1d0e8e683e5fdb27e12de53209d05a4a37e6246
50.67 Successfully built Flask-Mail Flask-Script sh rq-dashboard flasgger mysqlclient sgmllib3k
50.67 Failed to build cf-units
50.76 ERROR: Failed to build installable wheels for some pyproject.toml based projects (cf-units)
------
failed to solve: process "/bin/sh -c apt-get update &&     apt-get -y install libxml2-dev netcdf-bin rsync &&     cd /usr/local/src && pip install -U pip &&     pip install --no-cache -r /glider-dac/requirements.txt &&     apt-get -y remove libxml2-dev && rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 1

MathewBiddle, 2025-03-28:
@ocefpaf I heard about this tool that might help here? I think it's called `pixi` ? 😉

ocefpaf, 2025-03-28:
> [@ocefpaf](https://github.com/ocefpaf) I heard about this tool that might help here? I think it's called `pixi` ? 😉

Ben and I touched based on this yesterday. I'm debugging this today, there are some other issues related to updating the dependencies and code here though that will require us to do it n a way slow and incremental way.

benjwadams, 2025-04-03:
Tetra Tech might have additional restrictions on retrieving packages from `conda-forge`.  I discussed this some with Felipe before he codes up potential modifications to the Docker build system.

ocefpaf, 2025-04-03:
Thanks for the clarification Ben. We are waiting on more info from TetraTech and that decision. We are here to answer any question they may have BTW.

leilabbb, 2025-04-04:
UPDATE:

I got a cf-units error back when I completely removed Docker builds and started over.

**ERROR:**

 Building wheel for cf-units (pyproject.toml): finished with status 'error'
19.04   error: subprocess-exited-with-error
19.04   
19.04   × Building wheel for cf-units (pyproject.toml) did not run successfully.
19.04   │ exit code: 1
19.04   ╰─> [22 lines of output]
19.04       Traceback (most recent call last):
19.04         File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
19.04           main()
19.04         File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
19.04           json_out["return_val"] = hook(**hook_input["kwargs"])
19.04                                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19.04         File "/usr/local/lib/python3.11/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 280, in build_wheel
19.04           return _build_backend().build_wheel(
19.04                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19.04         File "/tmp/pip-build-env-xnzbu2x3/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 438, in build_wheel
19.04           return _build(['bdist_wheel', '--dist-info-dir', str(metadata_directory)])
19.04                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
19.04         File "/tmp/pip-build-env-xnzbu2x3/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 426, in _build
19.04           return self._build_with_temp_dir(
19.04                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
19.04         File "/tmp/pip-build-env-xnzbu2x3/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 407, in _build_with_temp_dir
19.04           self.run_setup()
19.04         File "/tmp/pip-build-env-xnzbu2x3/overlay/lib/python3.11/site-packages/setuptools/build_meta.py", line 320, in run_setup
19.04           exec(code, locals())
19.04         File "<string>", line 156, in <module>
19.04         File "<string>", line 69, in get_package_data
19.04       **ValueError: Require to set UDUNITS2_XML_PATH for a cf-units wheel build.**
19.04       [end of output]
19.04   
19.04   note: This error originates from a subprocess, and is likely not a problem with pip.
19.04   **ERROR: Failed building wheel for cf-units**
19.04   Building wheel for sgmllib3k (setup.py): started
19.14   Building wheel for sgmllib3k (setup.py): finished with status 'done'
19.14   Created wheel for sgmllib3k: filename=sgmllib3k-1.0.0-py3-none-any.whl size=6047 sha256=3b59361d1d0225fce7abfc109511d14e412e5a97464b7b141aa69defcee68271
19.14   Stored in directory: /tmp/pip-ephem-wheel-cache-y8gk0ogh/wheels/3b/25/2a/105d6a15df6914f4d15047691c6c28f9052cc1173e40285d03
19.14 Successfully built Flask-Mail Flask-Script sh rq-dashboard flasgger mysqlclient sgmllib3k
19.14 Failed to build cf-units
19.23 ERROR: Failed to build installable wheels for some pyproject.toml based projects (cf-units)
------
failed to solve: process "/bin/sh -c apt-get update &&     apt-get -y install rsync &&     pip install --no-cache -U pip &&     pip install --no-cache -r /glider-dac/requirements.txt &&     rm -rf /var/lib/apt/lists/*" did not complete successfully: exit code: 1

To get it to work I had to add these two lines to the Dockerfile:
ENV UDUNITS2_XML_PATH=/usr/share/xml/udunits/udunits2.xm
RUN ...  libudunits2-dev

So, it looks like on Apple Silicon M3 it won't do without the above

ocefpaf, 2025-04-04:
It should pull the wheel, something is amiss here. I'll try to build that on CIs next week to see if we can debug this.

benjwadams, 2026-02-19:
Leila was able to build `develop` branch.  I'll close this for now, but if there is further interest we can revisit adding ARM test runners to the CI setup.

closed 2026-02-19 by benjwadams