ioos/glider-dac #446: ERDDAP monitoring

benjwadams, 2025-04-30:
### What should we add?

At the DMAC meeting today, it was mentioned that ERDDAP can ship metrics through Prometheus.

We also have a script to write metrics for data ERDDAP is serving compared to FTP to InfluxDB here: https://github.com/ioos/glider-dac/blob/main/scripts/glider_check_update_erddap_influx.py

This likely would have helped detect some of the failure modes @jenseva experienced recently.

However, I think it would be good to port the script over to Prometheus, given that recent versions of ERDDAP now support a metrics endpoint.

benjwadams, 2025-05-16:
cc @pavaniankam92

sarinamann-noaa, 2025-05-19:
@benjwadams please provide updates to the Label, Type fields. If this is not mission critical, it should be backlogged.

benjwadams, 2025-11-04:
ELK stack has now been enabled for the web logs we are receiving on the Glider DAC server.

benjwadams, 2025-11-13:
Add Prometheus ERDDAP metrics to dashboards.