ioos/glider-dac #448: Data Upload Status Page for Providers

sarinamann-noaa, 2025-05-08:
### What should we add?

At the DMAC meeting, a user from PacIOOS expressed the need for a status page where users/providers can see real-time updates on the status of their data upload. Other users agreed.
Ideally, this would be a web-based platform where users can login or enter a unique ID for their data upload and see if any issues are flagged and if the dataset has been successfully uploaded and appears in ERDDAP.

benjwadams, 2025-05-16:
Will probably take shape as part of https://github.com/ioos/glider-dac/issues/412, which would likely be broken off or closed in favor of converting to a milestone for events occurring on the GDAC system.

Soft blocked by #424 -- We could still implement this, but it might further writing of code against an obsolete and probably insecure database layer and further reduplication of effort in the form of porting it over to the new DB layer.

sarinamann-noaa, 2025-05-19:
Feedback from @MathewBiddle - Add file size to IOOS Glider DAC Status page
https://github.com/ioos/glider-dac/issues/451#issue-3068941336

kerfoot, 2025-06-16:
Much of this information is available on the [status page](https://gliders.ioos.us/status/?cat=datasets&mode=rt&status=incomplete). Clicking on a glider will take you to a detailed summary of the data set status. For example, clicking on the [bass-20250601T0000 link](https://gliders.ioos.us/status/datasets/?dataset_id=bass-20250601T0000) takes you to a page that summarizes the data set status. In this case, you can see that there are potential issues with attribution and summary. These (or other issues) may be the reason that the data set is not showing up or showing up incorrectly.

Clicking on the [missing real-time active deployments](https://gliders.ioos.us/status/?cat=datasets&mode=rt&missing=True&status=all) displays alll data sets that are not currently showing up on ERDDAP. This is most likely due to issues with one or more NetCDF files. For each data set on this page, there is a checkmark if one or more NetCDF files has been uploaded to the registered data set. If you see a checkmark, the most likely cause of the missing data set is due to the NetCDF files themselves.

I have been working on a major upgrade to these pages that will provide full search functionality and highlight other potential issues with the data set. The page will be hosted at Rutgers University in order to maintain control over updates and in the event that the current data set status page is decommissioned. This will include, among other things, a report of how any NetCDF files have been uploaded compared to the number of profiles in the data set. Ideally, this is a 1:1 as each NetCDF file represents a single profile. 

The status page also has a simple [API ](https://gliders.ioos.us/status/api/) that allows users to programmatically retrieve information on each data set.

I'm still working on the updates and would be happy to include any additional information that might be useful to data providers in order to determine the status of the data set. I'm developing these pages on my own time but the goal is to provide the data providers with a detailed view on each data set that can potentially help them diagnose issues with data sets. Happy to discuss further if you'd like. I'm not looking for funding on this if that is a concern. The goal is to help data providers (including myself).