ioos/glider-dac #449: User request to include biodata in GDAC

sarinamann-noaa, 2025-05-08:
### What should we add?

At the Annual DMAC Meeting (April 29 - May 1, 2025) one user requested adding additional **biodata** fields to the GDAC ERDDAP server. 

*Sarina will double check user info/comments and provide more information

benjwadams, 2025-05-14:
I believe we already include variables which define a standard name.

sarinamann-noaa, 2025-05-15:
Correspondence sent to users plus a follow up. No responses yet. 

---------- Forwarded message ---------
From: Sarina Mann - NOAA Affiliate <sarina.mann@noaa.gov>
Date: Fri, May 9, 2025 at 8:57 AM
Subject: Glider DAC Variables- biogeochemical data
To: yui@mbari.org <yui@mbari.org>

Hi Yui,

At this year's DMAC meeting last week, we had a brief conversation about biogeochemical (BGC) data making its way into the GDAC. From last year's notes, I see a brief description that you asked about the status/process to include BGC parameters into the GDAC, including Oxygen, Chl, backscatter, pH, nitrate. As you know, we currently accept anything in the .nc file.
I was hoping that you could clarify this request. For example, did you want to know when QC would be applied to those variables (as right now QC is mainly applied to the primary variables - temperature, salinity...)? Are there other BGC variables that are crucial to your process? Do you have anything else to add to this request?

Although the GDAC is operating only at a mission-critical capacity at this time, we hope to continue gathering user input and adding to our backlog for development when funding is more secure.

---------- Forwarded message ---------
From: Sarina Mann - NOAA Affiliate <sarina.mann@noaa.gov>
Date: Fri, May 9, 2025 at 9:02 AM
Subject: Glider DAC - Biogeochemical Data
To: <hruhl@mbari.org>

Hi Henry,

At the DMAC meeting last week, I recall you had a question about biogeochemical (BGC) data (one other user has asked about oxygen, Chl, backscatter, pH, nitrate) making its way into the Glider DAC. 

Although the GDAC is operating only at a mission-critical capacity at this time, we plan to continue gathering user input and adding to our backlog for development when funding is more secure.

Would you mind expanding upon this request so I can accurately document it? For example, did you want to know when QC would be applied to those variables (as right now QC is mainly applied to the primary variables - temperature, salinity...)? Are there other BGC variables that are crucial to your process? Do you have anything else to add to this request?