ioos/glider-dac #451: Add file size to IOOS Glider DAC Status page

MathewBiddle, 2025-05-16:
### What should we add?

Would it be possible to add the physical size of the datasets to https://gliders.ioos.us/status/?cat=datasets&mode=all (add a column for `fileSize (MB)`? 

That would allow us to calculate some metrics on disk space needed/used for physical files which is helpful for data calls across NOAA.

### Reference

The following shell should give the response to populate the column with:
```
du -sh [dataset_name]
```

Assigning to @kbailey-noaa and @sarinamann-noaa to decide if it's something worth exploring.

benjwadams, 2025-05-16:
The requested feature enhancement pertains to a separate code base than the Glider DAC Providers application.  The DAC status application you are seeing was written by @kerfoot and possibly some others at Rutgers.  I'm not sure if the source code resides on GitHub, but we do have it on our server.  I believe part of it uses code from https://github.com/kerfoot/gdac-tools.  If I recall correctly, it pulls some of its information from the "old" DAC status application APIs.  This code is located at https://github.com/benjwadams/glider-dac-status/tree/plots_traj_hotfix and the ERDDAP we host at the DAC.  This is somewhat confusing, but basically the new UI for the GDAC status is some PHP code backed by fetch scripts in Python and shell scripts, which may in turn interact with some of the API endpoints we host.  In any event, you should already be able to obtain the information by navigating to a dataset on the status application and following to the "Source Files" link, albeit the information in file size is in bytes: 

E.g. https://gliders.ioos.us/status/datasets/?dataset_id=cp_376-20250330T1748 (follow link to "Source Files")-> http://gliders.ioos.us/erddap/files/ru39-20250423T1535/

Closing issue as out of repository scope.

closed 2025-05-16 by benjwadams

MathewBiddle, 2025-05-23:
Thanks @benjwadams for the context. I can use https://gliders.ioos.us/erddap/files/ to find the answers I need. Kudos to the team for having this accessible already.