ioos/glider-dac #457: Error on GDAC Map - osu266

sarinamann-noaa, 2025-07-07:
The osu266 glider is appearing with erroneous coordinates on the [GDAC map](https://gliders.ioos.us/map/) this morning. Please look into these coordinates as soon as you can, preferably before this afternoon's hurricane glider call- thank you!
I am assigning both Ben & Andy just to be thorough.

[image]

sarinamann-noaa, 2025-07-07:
Thanks for clarifying, Andy

ocefpaf, 2025-07-08:
> Bad trajectory being generated. https://gliders.ioos.us/status/summary/static/json/trajectories/mbari/OSU266-20250307T0000.json

Not sure if this helps, but that JSON seems fine. It is probably a plotting issue. See the same json on geojson.io:

[image]

benjwadams, 2025-07-09:
There's a QC process that eventually picks up extreme outliers like this.  Administrator input was not needed.