ioos/glider-dac #472: QC - GeoJSON generated trajectory

sarinamann-noaa, 2025-08-12:
From [GH issue #470 ](https://github.com/ioos/glider-dac/issues/470)

Wed, Aug 6, 2025 at 8:20 AM
Adams, Benjamin <[Benjamin.Adams@tetratech.com](mailto:Benjamin.Adams@tetratech.com)> wrote:
Location QC does have to run along with other QC, so sometimes artifacts like this still end up on the map. It may be worthwhile to run an outlier check for the generated GeoJSON tracks in addition to the QC. The location QC flagging is still valuable in its own right.

Tues, Aug 12, 2025 @benjwadams 
QC does not run instantaneously, so this can occur. If we want to filter at GeoJSON/generated trajectory level I'd suggest opening a separate, more focused issue.

benjwadams, 2025-08-20:
Related: https://github.com/benjwadams/glider-dac-status/pull/4

leilabbb, 2025-08-21:
A new filters were added to the trajectory.py to take out the bad coordinates  Here is the Pull request link https://github.com/benjwadams/glider-dac-status/pull/3 pending a merge to start effect.

leilabbb, 2025-09-19:
Recommend closing this issue:
The new [QC checks](https://github.com/benjwadams/glider-dac-status/pull/5) are in effect

leilabbb, 2025-10-02:
@benjwadams  please confirm that the land mass detection issue was addressed (maybe add the PR#) and I recommend closing this issue.

benjwadams, 2025-10-02:
I added a unit test and can confirm that the land mass tests are working performantly.

closed 2025-10-02 by benjwadams