ioos/glider-dac #475: sg678 missing the ERDDAP and THREDDS links in the GDAC Map

sarinamann-noaa, 2025-08-21:
---------- Forwarded message ---------
From: Kathleen Bailey - NOAA Federal <Kathleen.Bailey@noaa.gov>
Date: Mon, Aug 18, 2025 at 10:36 AM
Subject: Not urgent: GDAC map
To: _NOS Glider DAC Support <glider.dac.support@noaa.gov>

FYI, as I was troubleshooting something, I noticed SG678 (south of puerto rico) is missing the ERDDAP and THREDDS links in the GDAC Map.  These typically show up in the lower left when I click on a track.
k

-
Kathleen Bailey
Chief, Observation and Prediction Branch
U.S. Integrated Ocean Observing System

sarinamann-noaa, 2025-08-21:
---------- Forwarded message ---------
From: 'Adams, Benjamin' via _NOS Glider DAC Support <glider.dac.support@noaa.gov>
Date: Tue, Aug 19, 2025 at 12:18 PM
Subject: Re: Not urgent: GDAC map
To: Kathleen Bailey - NOAA Federal <Kathleen.Bailey@noaa.gov>, _NOS Glider DAC Support <glider.dac.support@noaa.gov>

Appears to be a fixed-height div — the links are still actually there, just not visible.  I have proposed a simple CSS fix to the team members who work on Glider map.

leilabbb, 2026-09-02:
@nguyandy is this resolved