ioos/glider-dac #477: Address WMO ID duplicate issues

sarinamann-noaa, 2025-08-28:
### Who is requesting this?

Internal GDAC, several GDAC users impacted

### What is being requested?

Several providers, including the Navy, are experiencing gliders with multiple WMO IDs assigned.

---------- Forwarded message ---------
From: Kathleen Bailey - NOAA Federal <Kathleen.Bailey@noaa.gov>
Date: Wed, Aug 27, 2025 at 11:54 AM
Subject: ATTN Re: WMO ID Request
To: Moretti, Donald <Donald.Moretti@tetratech.com>, glider.dac.support@noaa.gov <glider.dac.support@noaa.gov>

Hi Don, GDAC team,
We have another case of duplicate WMO IDs, this time for Navy gliders.

NG288 now has 2 WMO IDs.  This glider already had an ID assigned in ~2018: https://gliders.ioos.us/erddap/info/ng288-20180801T0000/index.html

Same for NG665:
Here is the original ID: https://gliders.ioos.us/erddap/info/ng665-20220713T0000/index.html

I didn't check others.

Is there a script that can be run to show us which gliders (by call sign) have multiple WMO IDs?

Kathy

**From GDAC Meeting 08/28/25:**

**Process Issues:**
- When people request WMO ID, we use provider-given name for the glider. As of now, users can create a deployment through providers app and assign the WMO ID. The provider can modify the WMO ID manually or update the file. The NetCDF resubmittals (with changed metadata) won’t impact the existing WMO ID. Thus, a glider could have 2 WMO IDs- one from the NetCDF file and one from the system.
- There is a need for a constraint on the front-end to prevent this. Ideally an error would show up when a request is made that would yield a duplicate. 
- Process for correcting duplicate WMO IDs- ask providers to submit correct metadata. This can also happen on the back-end, but preference is for provider to change data.
- It is recommend that we only have one way to assign/enter WMO IDs. 
- The current documentation doesn’t ask provider to provide WMO ID, it just asks for it to be in metadata. The WMO txt file is part of outdated workflow; this shouldn’t be referenced anywhere in documentation. 
- 

**Work to be done immediately**
- Ben Adams has generated a query showing duplicate WMO IDs
- Requesting that the GDAC take care of the issue of duplicate WMO IDs by correcting the WMO ID fields in the metadata for the affected gliders. Going forward, we will adjust process to enforce correct WMO assignment. 
Action: Ben will clean up the output from his script (the duplicates), then the GDAC Team will enforce the workflow going forward.
Action: Ben provide the list of duplicates to Sarina, who will identify the specific glider datasets that need to be corrected, and will supply that list of changes to GDAC Support to change

### How would this request improve your current workflow?

Reduce confusion for providers who experience gliders with duplicate WMO IDs

### How can we get in contact with you to follow up on this request?

Internal GDAC- use GitHub

leilabbb, 2026-02-26:
Update:

> For Rutgers, Lori Garzio is the point of contact - email sent.
> 
> 4801927,rutgers,ud_134
> 4801927,rutgers,ud_476

The WMO ID for ud_134 should be 4801947

leilabbb, 2026-09-02:
Closing issue:

The providers app provides a note about the "Assigned WMO ID" if it exists or is used by more than one glider ([example](https://gliders.ioos.us/providers/deployment/maracoos_06-20250116T1522?username=none)). 

There is also a "Search by WMO ID" on the providers page that checks for the gliders in the system.

closed 2026-09-02 by leilabbb