ioos/glider-dac #479: Stored WMO IDs are fairly unclear

benjwadams, 2025-08-29:
### Briefly describe the issue

Stored WMO IDs in the database are based upon what the user creates when making the initial deployment or submitting a file named "wmo_id.txt" in the deployment, *NOT* what is submitted in netCDF files.  This leads to confusing situations where the reported WMO ID on the providers application differs from what is on ERDDAP and submitted downstream to NCEI.   The WMO IDs should reflect the contents of the most recently submitted netCDF file.

benjwadams, 2025-09-02:
This is a potentially separate, but related issue.  I'd recommend cross-referencing this issue from #477.

sarinamann-noaa, 2026-02-19:
More exploration needed- moved to to-do for next steps

sarinamann-noaa, 2026-03-24:
@benjwadams @leilabbb 
Is this something we would need to suggest that the providers help with? Do we want to ask the providers to review their file names or is this something we would adjust on our end?